bouffalolab/bouffalo_sdk #147: Missing BL702 HBN XCLK initialization in SDK v2.0

mdednev, 2023-05-22:
Hello,

Newest SDK 2.0 lacks HBN_Set_XCLK_CLK_Sel(...) calls to initialize HBN high-speed clock for BL702. Bluetooth LE and other wireless interfaces couldn't perform properly without such initialization, because they should use very stable XTAL 32M clock.
Older bl_mcu_sdk V1.4.5 has such calls, current bl_iot_sdk version has such calls too. 
This initialization should be added to the BSP's `system_clock_init(void)` function, so `system_clock_init(void)` should look like:
```C
static void system_clock_init(void)
{
    GLB_Set_System_CLK(GLB_DLL_XTAL_32M, GLB_SYS_CLK_DLL144M);
    GLB_Set_MTimer_CLK(1, GLB_MTIMER_CLK_BCLK, 71);
    HBN_Set_XCLK_CLK_Sel(HBN_XCLK_CLK_XTAL);
}
```

sakumisu, 2023-05-24:
Right, wireless need external 32M xtal, but currently wireless does not support for bl702.

sakumisu, 2023-05-25:
Thanks, you can submit pr to modify this.

mdednev, 2023-05-25:
Done. Created https://github.com/bouffalolab/bouffalo_sdk/pull/149

closed 2023-05-25 by mdednev