bouffalolab/bouffalo_sdk #148: BL616/BL618 and BL808C/BL808D (Bluetooth Classic/Zigbee/Proprietary 2.4GHZ)

techlabz, 2023-05-24:
@sakumisue  These are amazing chips that have been developed, glad to have stumbled across them. However, it seems like heavy documentation and development are missing of these chips.

Is there any particular reason wireless support isn't supported yet on BL808 ? There is virtually no mention of this in the reference manual.

Diagram seems to mention support for Bluetooth Classic for BL616/618 why hasn't that been implemented?

Is there support for Zigbee and other custom proprietary 2.4 GHZ stacks? Are the any licensing or mac/radio issues that are being faced right now? Is there a guarantee that these features will be supported in the future? Or is all the underlying registers available to develop ourselves?

sakumisu, 2023-05-24:
BL808 wireless code depends on iot sdk too much, merging into bouffalo sdk is a big and hard work, and we have no time to support it, thanks.

We have refer manual and datasheet, no missing. All wireless doc in NDA, no access to give for you.

gamelaster, 2023-05-24:
@sakumisue what about BL616 BT Classic support in MCU SDK?

Thanks :)

sakumisu, 2023-05-24:
> @sakumisue what about BL616 BT Classic support in MCU SDK?
> 
> Thanks :)

No plan to support in current sdk.

netmonk, 2023-05-27:
Is it because of propretary driver for radio stacks ? I mean theses mcu are fairly better of any competitor hardware, beating esp32-c3/6 in any field, but at least radio are open access in ardiuno/esp-idf sdk for those devices. 

Not having access without an NDA (which mean having a company and an official project which must be closed source to respect NDA)  is a show stopper for many enthousiast who would freely spend time to extend the code base for such new devices. 

ESP32 succeeded to grab some market share from ARM cause they release all doc especially radio driver and code which is a big winner compared to ARM missing such capabilities on their hardwares. Wonder why ESP saw their codebase increase to large extent remplacing ARM successfully over the year in many field where ARM has no competition since many years. And sadly i guess their riscv will have lot of support from community enthousiast and professionnal for the same reasons. 

from my perspective keeping radio code access  behind an NDA and not giving possibility for people to release their own code under open source if they use radio code under signed NDA will strongly limit the adoption of your risc platform by the public, amator and enthousiast. Therefore no need to sells them on aliexpress cause they are pretty useless peace of silicon under such circumstances. 

And we also should sue Sipeed for advertising wifi/bluetooth and zigbee when selling those devices without explaining that those are very liimted and restricted by default. Which indeed look like fake advertising.

nickfox-taterli, 2023-05-28:
> Is it because of propretary driver for radio stacks ? I mean theses mcu are fairly better of any competitor hardware, beating esp32-c3/6 in any field, but at least radio are open access in [...]

I very much agree with you, in fact, not only the wireless function is not supported, other documentation is also quite lacking, and no one is even willing to give you this feature is why so do. I recognize that like Nordic also only open SoftDevice binaries, but, at least let me use the wireless function up, it is very important.

gamelaster, 2023-05-28:
@netmonk @nickfox-taterli I totally understand you guys, and also agree with some of your statements.

Please note, that source code for most of the radio stuff are always under NDA. I am not sure how much Espressif released, but probably not much. Only exception is Nordic, which released registers for their BLE cores.

Anyway, even the situation with documentation, support and SDK is not perfect from Bouffalo, but I am playing and watching Bouffalo Lab and their chips from 2020, and I am happy with the things, they provide us. It is not the best, but better than many other Chinese AIoT chip vendors. At least for BL602 and BL702 they published some documentation, and quite a lot code for the chips, in quite good shape, so it is useful, but not easy for beginners or tinkerers.

Although, I agree with you and I would like to see more things supported by Bouffalo SDK, although, that's all upon Bouffalo decision.

nickfox-taterli, 2023-05-28:
> @netmonk @nickfox-taterli I totally understand you guys, and also agree with some of your statements.
>
> Please note, that source code for most of the radio stuff are always under NDA. I am not sure [...]

It is understandable that the source code is not provided for confidentiality purposes, but at least it should be available lib, sometimes we do not necessarily care what the source code is, there are many very interesting (although not open source) AIoT chips in China, such as the CH579 provided by WCH, although his wired network and Bluetooth do not provide any code, only the operating library, providing a part of the registers. But this is good, can be used up.

Bouffalo Lab's documentation is not particularly good (probably due to my reading ability), sometimes even a glance can not find the JTAG debug interface where, can not understand the three kernel boot process and how to store the firmware, of course, I know someone has solved this problem, such as rt-thread, but I still do not understand.

There are some issues will not get the official response, there are some documents are out of date, as well as the initial hard to get started debugging, many software only Chinese, I can understand, is currently trying to merge mcu-sdk and iot-sdk, so there are many many 😂, but I hope to do a good job.

gamelaster, 2023-05-28:
@nickfox-taterli if you are interested in more information about BL808 Linux development made by community, check out https://openbouffalo.org/index.php/Main_Page#Join_us and join our chat platforms :blush:

nickfox-taterli, 2023-05-28:
> @nickfox-taterli if you are interested in more information about BL808 Linux development made by community, check out https://openbouffalo.org/index.php/Main_Page#Join_us and join our chat platforms 😊

It's a pity that I searched for your content earlier, but it doesn't support Sipeed M1S (BL808), and of course, the wireless function and LP don't seem to be supported either.

But it is a very good project.

gamelaster, 2023-05-28:
@nickfox-taterli we support Sipeed M1s, and we got LP working. WiFi is very early in progress.

netmonk, 2023-05-28:
> @netmonk @nickfox-taterli I totally understand you guys, and also agree with some of your statements.
>
> Please note, that source code for most of the radio stuff are always under NDA. I am not sure [...]

Most of the time the radio code might be under NDA. Ok for that, but at least when i use esp-idf or arduino for esp32, it doesnt require an NDA to use wifi or BLE in a matter of minute. That's the point and this explain why this hardware succeed to grab market share against ARM supremacy. 

Lesson must be learnt from them.

sakumisu, 2023-05-29:
> Most of the time the radio code might be under NDA. Ok for that, but at least when i use esp-idf or arduino for esp32, it doesnt require an NDA to use wifi or BLE in a matter of minute. That's the [...]

No, using wifi or ble or providing wifi/ble doc is two thing not the same, thanks.

sakumisu, 2023-05-29:
If you want to use bl808 wifi, please refer to sipeed sdk not bouffalo_sdk.

iamdavidcz, 2023-05-29:
@sakumisu
> We have no time

This seems to be the most often excuse from this Chinese silicon companies... so why not publish the register level docs for WiFi and BLE so the community could figure out the missing pieces instead of relying on you not having time *all the time*?

sakumisu, 2023-05-29:
> @sakumisu
> 
> > We have no time
> 
> This seems to be the most often excuse from these Chinese silicon companies... so why not publish the register level docs for WiFi and BLE so the community could figure out the missing pieces instead of relying on you not having time _all the time_?

Do you know what is license？please do not play a joke.

gamelaster, 2023-05-29:
@iamdavidcz 
> This seems to be the most often excuse from these Chinese silicon companies...
Yes, although, from my personal experience with Bouffalo and with embedded business, they really don't have that much time, and also getting good developers nowadays even in China is problem. We should be really happy for support and responses we are getting here. As I mentioned multiple times, it is not perfect, but better than other companies.

> so why not publish the register level docs for WiFi and BLE so the community could figure out the missing pieces instead of relying on you not having time all the time?

As it was mentioned upper, and also confirmed by sakumisu, all RF and PHY related stuff are strictly under NDA, it is not Bouffalo's fault, but rather IP core vendors, which are too sensitive to release any documentation for their stuff, in terms of RF, it is company named Ceva.

So basically, we have BL808 WiFi (and as far as I remember) BLE blobs, just integrated in IoT SDK. It can be pulled out and re-used. ZigBee (as whole thing) is under NDA. Matter or Thread might pop up someday, but I know one of those are under NDA too.

When we are looking retrospectively, Bouffalo provided WiFi & BLE blobs for all of their chips, BL602 and BL702. Also, someone leaked Zigbee blobs for BL702. New chips BL61X series have support of WiFi & BLE directly in this SDK, what is amazing.
So in the end, it is not that bad by my opinion.

iamdavidcz, 2023-05-29:
@gamelaster
>  As it was mentioned upper, and also confirmed by sakumisu, all RF and PHY related stuff are strictly under NDA, it is not Bouffalo's fault, but rather IP core vendors, which are too sensitive to release any documentation for their stuff, in terms of RF, it is company named Ceva.

Thanks for the elaborate answer, I wasn't quite aware of the whole situation. However, in my opinion it should be stated clearly in README, so people could understand the situation clearly...

nickfox-taterli, 2023-05-29:
I hope to be able to use the wireless function of BL808 simply as soon as possible (with this combine sdk), and of course, there is little presence of the LP core.

UncleGrumpy, 2023-05-29:
> Do you know what is license？please do not play a joke.

So I am still confused, you are contacting yourself... First you say no time (which leaves a shred of hope that someday this will be a useful device), and then when a comment is made about how long it is taking, you say it is because of licensing problems (which shatters any hope of doing anything useful). Can you at least clarify:
Will we *ever* be able to use bl808 WiFi with Buffalo_SDK?

sakumisu, 2023-05-29:
> So I am still confused, you are contacting yourself... First you say no time (which leaves a shred of hope that someday this will be a useful device), and then when a comment is made about how long [...]

I think you misunderstand, wifi ble registers and doc are licensed, we just have no time to merge 808  wifi from iot sdk to bouffalo sdk because it depends too much components, thanks, i dont want to say again.

techlabz, 2023-05-29:
@sakumisu 
It does seem like this is a reoccurring theme, you have all this information, but there is no mention of this anywhere in the documentation. What is under NDA and what is not, this product is being advertised as a BT/BLE, Wifi, Zigbee, 2.4GHZ solution but as of now it is hard for someone to integrate. In my opinion significant amount of time needs to spent on updating the documentation pointing developers in the right direction, the iot-sdk doesn't mention that it can be used for BL808 for example. The reference manual doesn't mention Wireless functionality is under NDA, and who to contact. Not sure we can use this product as of now will have to stick with Nordic/Realtek, but will check back in a few months to see where progress is and hopefully evaluate for future products.

nickfox-taterli, 2023-05-30:
> @sakumisu It does seem like this is a reoccurring theme, you have all this information, but there is no mention of this anywhere in the documentation. What is under NDA and what is not, this product [...]

Because this is a wireless chip with temporarily imperfect wireless functionality, it is indeed a joke, I do not know what is more important than porting a major function, unless the wireless function is not actually the main function of this chip.

sakumisu, 2023-05-30:
> Because this is a wireless chip with temporarily imperfect wireless functionality, it is indeed a joke, I do not know what is more important than porting a major function, unless the wireless [...]

Do not play a joke, why not use iot sdk? It has bl808 wifi.

nickfox-taterli, 2023-05-30:
> Do not play a joke, why not use iot sdk? It has bl808 wifi.

So, either you use the SDK that can operate the hardware well, or you use WiFi, please try not to use peripherals.

I'm not kidding, I just hope that the integration of the SDK can do better, ESP8266, RTOS_SDK also do not do very well, but to esp-idf, everything becomes good, I do not know what is no time, or that the development of their own company's software belongs to have free time to do things.

See what others say.

gamelaster, 2023-05-30:
@nickfox-taterli you can use IoT SDK for BL808, actually, IoT SDK supports more BL808 peripherals than current Bouffalo SDK. There is BL808 branch in official iot sdk, or you can use more complete one from SiPeed. Only piece missing is Zigbee. So where is issue?

sakumisu, 2023-05-30:
> So, either you use the SDK that can operate the hardware well, or you use WiFi, please try not to use peripherals.
>
> I'm not kidding, I just hope that the integration of the SDK can do better, [...]

OK，你应该懂中文，我想说的是，iot sdk里面 808 wifi 依赖了几百个组件，如果移植过来，基本上等于把iot sdk都拿过来，这个工作你觉得很简单吗？其次，做这件事，你提供人力？还是提供money？

netmonk, 2023-05-30:
So to summerise can anyone can list what device wifi/ble/zigbee is supported for wich hardware in which SDK ? 

I have Sipeed board with bl616 and bl702, i dont see any support in m0s_bl616_exemple nor in M0sense_bl702_exemple...

sakumisu, 2023-05-30:
> So to summerise can anyone can list what device wifi/ble/zigbee is supported for wich hardware in which SDK ?
> 
> I have Sipeed board with bl616 and bl702, i dont see any support in m0s_bl616_exemple nor in M0sense_bl702_exemple...

Can you see readme?

gamelaster, 2023-05-30:
@netmonk 
For BL616, you can use this SDK, as it supports both WiFi & BLE. Additionally, you have examples from Sipeed here: https://github.com/sipeed/M0P_BL618_examples

For BL702, you can use https://github.com/bouffalolab/bouffalo_sdk/tree/release_v1.4.5 which completely supports BL702 and it's BLE stack. It works properly, as we use it in Pinecil V2.

sakumisu, 2023-05-30:
> So to summerise can anyone can list what device wifi/ble/zigbee is supported for wich hardware in which SDK ?
> 
> I have Sipeed board with bl616 and bl702, i dont see any support in m0s_bl616_exemple nor in M0sense_bl702_exemple...

List on Readme, you can see it.

nickfox-taterli, 2023-05-31:
Your company's main product has no time and no money probably all invested in marketing.

dreamcmi, 2023-05-31:
@nickfox-taterli First of all, since you replied in English, I will also reply to you in English. As I have mentioned above, the official team is working on porting the project, but because the old project has too many dependencies, the progress will be slow. This involves the workload and confidentiality issues, and it does not mean that they will not provide it. Besides, you can also choose the iot-sdk provided by sipeed to develop wireless services. If you are really in a hurry, why don’t you pay for a solution? I believe that the project you are anxious about must be very profitable. Compared with Espressif, their initial progress was also very slow, and there even were projects where esp8266 was not open source and enthusiasts reverse engineered it to implement functions. Don’t think that everything is owed by the manufacturer, they need time, manpower, and funds to maintain open source. Open source is a very energy-consuming work, and if there is no stable resource support, it will not last for a long time. Espressif has proved this point, they have devoted a large part of their manpower and funds to open source.

closed 2023-05-31 by sakumisue