bouffalolab/bouffalo_sdk #153: BL808 pine ox64 Fat type fs ram compile issue

davidawesome02, 2023-06-17:
After direct clone from github current version, I cannot compile the Fat example at all.

Command:
```
make CHIP=bl808 BOARD=bl808dk CPU_ID=m0
```

System specification:
Using a rockpro64 from pine64 via box64 to run all software, no errors from box64 are going on.... and some other examples work.
(That was a lot of 64)

My output was:
```
<path here>/github_clones/toolchain_gcc_t-head_linux/bin/../lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: build_out/fatfs_bl808_m0.elf section `.bss' will not fit in region `ram_memory'
<path here>/github_clones/toolchain_gcc_t-head_linux/bin/../lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: region `ram_memory' overflowed by 61468 bytes
collect2: error: ld returned 1 exit status
make[3]: *** [CMakeFiles/fatfs_bl808_m0.elf.dir/build.make:107: build_out/fatfs_bl808_m0.elf] Error 1
make[2]: *** [CMakeFiles/Makefile2:529: CMakeFiles/fatfs_bl808_m0.elf.dir/all] Error 2
make[1]: *** [Makefile:91: all] Error 2
```

I didn't change any files or anything after downloading, so I have no idea how this happened.

Posting this in case anyone knows how to fix it...

I will update if I find a fix...

davidawesome02, 2023-06-17:
Update; its a bug with a file in 
```
bouffalo_sdk/bsp/board/bl808dk/board.c.o
bouffalo_sdk/bsp/board/bl808dk/fw_header.c.o
```

not components, or the example code

davidawesome02, 2023-06-17:
nevermind, something else in 
```
bouffalo_sdk/bsp/board/bl808dk/
```

neither of those sorry

sakumisu, 2023-06-19:
Reduce RW_Buffer in fatfs_demo by yourself, only 24K ram for users, and left is for wifi.