bouffalolab/bouffalo_sdk #159: Sdk having components that have never worked to the public listed as working

davidawesome02, 2023-06-25:
Hey, just leaving this, because this is kinda dumb....

All the btble examples are broken on bl808, with 2.0.0 sync saying they dont work, then the latest sync saying it dose, only one problem, it is missing the files required to work at all...

Also, exfat is listed as a option in comments on the fatfs examples, and it too has a invalid partition table, and thats only if you get the sdk working with a old version....

I know this is a lot to ask, but can you at least update the files to say that it will not work, or actually fix it...

Thanks, and sorry for being this way, but it just sucks that I have a bl808 and cant even utilize it properly with broken examples that don't even have old working versions, seemingly working at least at one time during development.

And if anyone has any patches or anything for it that would be cool, I will prob look soon to see if I can fix it and submit a pull, but I am kinda new to C, so lets see what happens

Thanks

sakumisu, 2023-06-28:
Not work for bl808, thanks.

sakumisu, 2023-06-28:
exfat works.
```
[0m[I][MAIN] FileSystem cluster size:16-sectors (8192-Byte)
[0m[I][MAIN] 
******************** be about to write test... **********************
[0m[I][MAIN] Write Test Succeed! 
[0m[I][MAIN] Single data size:32768 Byte, Write the number:1024, Total size:32768 KB
[0m[I][MAIN] Time:7379ms, Write Speed:4440 KB/s 
[0m[I][MAIN] 
******************** be about to read test... **********************
```

davidawesome02, 2023-07-05:
> exfat works.
>
> ```
> �[0m[I][MAIN] FileSystem cluster size:16-sectors (8192-Byte)
> �[0m[I][MAIN]
> ******************** be about to write test... **********************
> �[0m[I][MAIN] Write Test Succeed! [...]

can you send your config and stuff? I had a issue where it would succeed but if the wrong config was used it would lie and make a different fat type file system

and even if it did make it, can your computer access it?

sorry, but I have been working on this, and I have a feeling this may be just like my other attempts that failed for no reason...

Thanks!

closed 2023-08-28 by sakumisue