bouffalolab/bouffalo_sdk #161: BL616 helloworld no console

berkutta, 2023-07-04:
I'm trying to get the helloworld example running on an Ai-M62-12F-Kit. I can build and flash it, but I don't get any console output via picocom.

Building and flashing:
```
make CHIP=bl616 BOARD=bl616dk
make CHIP=bl616 BOARD=bl616dk flash COMX=/dev/ttyUSB0
```

No output even after pressing RESET multiple times:
```
sudo picocom -b 2000000 /dev/ttyUSB0
picocom v3.1

port is        : /dev/ttyUSB0
flowcontrol    : none
baudrate is    : 2000000
parity is      : none
databits are   : 8
stopbits are   : 1
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : no
hangup is      : no
nolock is      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv -E
imap is        : 
omap is        : 
emap is        : crcrlf,delbs,
logfile is     : none
initstring     : none
exit_after is  : not set
exit is        : no

Type [C-a] [C-h] to see available commands
Terminal ready

Terminating...
Skipping tty reset...
Thanks for using picocom
```

Compiler version:
```
riscv64-unknown-elf-gcc --version 
riscv64-unknown-elf-gcc (Xuantie-900 elf newlib gcc Toolchain V2.6.1 B-20220906) 10.2.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

gamelaster, 2023-07-04:
Since you can flash program with same UART, pins should be correct. Did you tried another baudrates, like 115200?

berkutta, 2023-07-04:
I have tried 9600, 115200 and 2000000. I think 2000000 should be correct https://github.com/bouffalolab/bouffalo_sdk/blob/master/bsp/board/bl616dk/board.c#L196

berkutta, 2023-07-04:
Ok, it works with an external USB-UART converter connected to the Pins TX and RX on the Pinheader. This means no SDK issue, but issue on the AiThinker board..

closed 2023-07-04 by berkutta