bouffalolab/bouffalo_sdk #162: Building for BL618

berkutta, 2023-07-10:
How can I build for BL618? Do I just need to specify CHIP=bl616?

Getting errors about non-existing files if I specify CHIP=bl618.

```
➜  helloworld git:(master) ✗ make CHIP=bl618
./../../tools/cmake/bin/cmake -S . -B build -G "Unix Makefiles" -DCROSS_COMPILE=riscv64-unknown-elf- -DCHIP=bl618 -DCPU_ID= -DBOARD=bl616dk -DBOARD_DIR= -DCONFIG_DEBUG=y -DCONFIG_ROMAPI=y -DCONFIG_USB_HS=y -DCONFIG_COMX=COM5 -DCMAKE_EXPORT_COMPILE_COMMANDS=OFF -DCONFIG_TLSF=y
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/user/prog/bouffalo/toolchain_gcc_t-head_linux/bin/riscv64-unknown-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/user/prog/bouffalo/toolchain_gcc_t-head_linux/bin/riscv64-unknown-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/user/prog/bouffalo/toolchain_gcc_t-head_linux/bin/riscv64-unknown-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- [register library : libc], path:/home/user/prog/bouffalo/bouffalo_sdk/components/libc
-- [register library : mm], path:/home/user/prog/bouffalo/bouffalo_sdk/components/mm
-- [register library : utils], path:/home/user/prog/bouffalo/bouffalo_sdk/components/utils
-- [register library : lhal], path:/home/user/prog/bouffalo/bouffalo_sdk/drivers/lhal
-- [register extern library : libcsi_xt900p32f_dsp], path:/home/user/prog/bouffalo/bouffalo_sdk/drivers/lhal
-- [register extern library : libpka], path:/home/user/prog/bouffalo/bouffalo_sdk/drivers/lhal
CMake Error at /home/user/prog/bouffalo/bouffalo_sdk/CMakeLists.txt:6 (add_subdirectory):
  add_subdirectory given source "drivers/soc/bl618/std" which is not an
  existing directory.

-- Configuring incomplete, errors occurred!
See also "/home/user/prog/bouffalo/bouffalo_sdk/examples/helloworld/build/CMakeFiles/CMakeOutput.log".
make: *** [../../project.build:72: build] Error 1
```

sakumisu, 2023-07-10:
Can you read readme? 618 is the same with 616

berkutta, 2023-07-10:
Ok, thanks

closed 2023-07-10 by berkutta

mrhempman69, 2023-11-29:
This is not in the readme dude