bouffalolab/bouffalo_sdk #163: No UART Output in wifi_mqtt_pub example

berkutta, 2023-07-10:
Hello

I'm trying to get the `wifi_mqtt_pub` example running on the BL616. It builds fine, and I can flash it. But I don't get any output via Terminal.

Despite that, I get the output via Terminal for the `helloworld` example. I don't know what the difference is.

```
make CHIP=bl616 BOARD=bl616dk
make CHIP=bl616 BOARD=bl616dk flash COMX=/dev/ttyUSB0
sudo picocom -b 2000000 /dev/ttyUSB0
```

gamelaster, 2023-07-10:
did you tried different baudrate?

berkutta, 2023-07-10:
Haven't found anything which should make the baudrate different between `wifi_mqtt_pub` and `helloworld`. But will check with the scope.

berkutta, 2023-07-10:
Looks like there is no communication on the TXD pin.

`wifi_mqtt_pub` example:
[image]

`helloworld` example:
[image]

sakumisu, 2023-07-11:
you must use bl618, mqtt must use psram

sakumisu, 2023-07-11:
So if you use bl616, it will crash in psram init.

berkutta, 2023-07-11:
Thanks, `wifi_tcp` example works :+1:

closed 2023-07-11 by berkutta