bouffalolab/bouffalo_sdk #167: USB host example does not work on M0S/BL616

harbaum, 2023-08-07:
The USB host demo won't even boot at all and gives no output on the UART.

The USB client demos all _do_work and also the usbhost exmaple from https://github.com/sipeed/M0S_BL616_example/tree/main/examples/peripherals/usbhost does work. But the one from this repository does nothing visible at all and doesn't even give the typical boot message.

I would like to test the current version from boufalolab since the the (old) version from sipeed fails most of the time:

```
[I/USB] EHCI HCIVERSION:0100
[I/USB] EHCI HCSPARAMS:000001
[I/USB] EHCI HCCPARAMS:0006
[I/USB] New low-speed device on Hub 1, Port 1 connected
[I/USB] New device found,idVendor:0e8f,idProduct:2517,bcdDevice:0100
[I/USB] The device has 2 interfaces
[I/USB] Enumeration success, start loading class driver
[I/USB] Loading hid class driver
[I/USB] Ep=81 Attr=03 Mps=8 Interval=10 Mult=00
[I/USB] Register HID Class:/dev/input0
[I/USB] Loading hid class driver
do not find /dev/ttyACM0
[I/USB] Unregister HID Class:/dev/input0
[E/USB] Port 1 enumerate fail
```

sakumisu, 2023-08-08:
From your log, it works.

harbaum, 2023-08-08:
> From your log, it works.

```
[I/USB] Loading hid class driver
[I/USB] Unregister HID Class:/dev/input0
[E/USB] Port 1 enumerate fail                     <---------------------
```
No, it doesn't work. This "fail" happens 95% of the time and the device doesn't work afterwards (keystrokes aren't reported).

harbaum, 2023-08-08:
> And if you use m0s/m0p, you must use sipeed sdk, thanks.
You are not supporting 3rd party boards? That's a problem for a developer like me.

You may reconsider that. E.g. Espressif gives great support for 3rd party hardware, and I am pretty sure that's one of the key factors of their success.

The Sipeed SDK is outdated and hasn't been updated since months. And since the M0S isn't much more than your BL616 with just a few passive components around it, I don't understand why the demos from your SDK wouldn't run there.

harbaum, 2023-08-08:
To answer my own question:

That example has PSRAM enabled and won't work without PSRAM. The example doesn't seem to need PSRAM and disabling it in `proj.conf` makes the example run on the M0S.

And it's definitely more reliable than the version in the SiPeed SDK.

closed 2023-08-08 by harbaum