bouffalolab/bouffalo_sdk #168: How to use SDIO2 on bl616

trevor403, 2023-08-15:
It looks like the bl602 exclusively used sdu as an SDIO2 slave interface. However `bl616_sdu.c` in the SDK looks like a stub when compared with the `[bl_iot_sdk_matter](https://github.com/bouffalolab/bl_iot_sdk_matter)` implementation.

Is the correct way to use the SDIO2 slave peripheral via `bflb_sdio2.c` and will there be an example like the `netbus_mgmr` component?

sakumisu, 2023-08-16:
SDIO2 is special and need private sdh host driver, not for public

trevor403, 2023-08-19:
I am more interested in `sdu` and how to use it on `bl616` as an SDIO Slave device.

trevor-clickup, 2023-09-05:
Is it true that SDIO2 in Slave mode will never be publicly supported by the `bl616`

trevor403, 2023-09-27:
So is this a private proprietary feature? Is it not actually offered and the marketing for this IC is incorrect? I am interested to know [Bouffalo Lab](https://en.bouffalolab.com/)'s response.

closed 2024-02-07 by sakumisue

gamelaster, 2024-02-07:
https://github.com/bouffalolab/bouffalo_sdk/tree/master/examples/peripherals/sdio2/sdio2_test