bouffalolab/bouffalo_sdk #170: SD card read error on BL616

harbaum, 2023-08-18:
I have been testing the example/fatfs on a BL616 and cleaned it up a little bit (see PR https://github.com/bouffalolab/bouffalo_sdk/pull/169). While most changes are just cosmetics (no more \0 bytes in ASCII test file) I added a small section that fills the receive buffer with $55 bytes before reading a new test data chunk.  I did this because all chunks contain the same data and it makes sense to invalidate this before reading and comparing the same data again.

To my very surprise the test now fails. Some data is not overwritten on read. 32 Bytes are still $55 after reading the second chunk:

```
  ____               __  __      _       _       _     
 |  _ \             / _|/ _|    | |     | |     | |    
 | |_) | ___  _   _| |_| |_ __ _| | ___ | | __ _| |__  
 |  _ < / _ \| | | |  _|  _/ _` | |/ _ \| |/ _` | '_ \ 
 | |_) | (_) | |_| | | | || (_|q | | (_) | | (_| | |_) |
 |____/ \___/ \__,_|_| |_| \__,_|_|\___/|_|\__,_|_.__/ 

Build:08:50:01,Aug 18 2023
Copyright (c) 2022 Bouffalolab team
======== flash cfg ========
flash size 0x00400000
jedec id     0xC86016
mid              0xC8
iomode           0x04
clk delay        0x01
clk invert       0x01
read reg cmd0    0x05
read reg cmd1    0x35
write reg cmd0   0x01
write reg cmd1   0x01
qe write len     0x02
cread support    0x01
cread code       0x20
burst wrap cmd   0x77
===========================
dynamic memory init success, ocram heap size = 327 Kbyte 
sig1:ffffffff
sig2:0000f32f
cgen1:9f7ffffd
[0m[I][MAIN] FileSystem cluster size:8-sectors (4096-Byte)

[0m[I][MAIN] ******************** be about to write test... **********************
[0m[I][MAIN] Write Test Succeed! 
[0m[I][MAIN] Single data size:32768 Byte, Write the number:1024, Total size:32768 KB
[0m[I][MAIN] Time:6172ms, Write Speed:5309 KB/s 

[0m[I][MAIN] ******************** be about to read test... **********************
[0m[I][MAIN] Read Test Succeed! 
[0m[I][MAIN] Single data size:32768Byte, Read the number:1024, Total size:32768 KB
[0m[I][MAIN] Time:1743ms, Read Speed:18799 KB/s 

[0m[I][MAIN] ******************** be about to check test... **********************
[0m[I][MAIN] Check Test Error! 
[0m[I][MAIN] Data Error!  Num:1/1024, Byte:17984($4640)/32768
00000000  49 27 76 65 20 62 65 65 6e 20 72 65 61 64 69 6e  I've been readin
00000010  67 20 62 6f 6f 6b 73 20 6f 66 20 6f 6c 64 20 0d  g books of old .
...
00004620  0a 20 20 20 20 54 68 65 20 6c 65 67 65 6e 64 73  .    The legends
00004630  20 61 6e 64 20 74 68 65 20 6d 79 74 68 73 20 0d   and the myths .
00004640  55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55  UUUUUUUUUUUUUUUU    <------------------
00004650  55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55  UUUUUUUUUUUUUUUU    <------------------
00004660  20 48 65 72 63 75 6c 65 73 20 61 6e 64 20 68 69   Hercules and hi
00004670  73 20 67 69 66 74 73 20 0d 0a 20 20 20 20 53 70  s gifts ..    Sp
...
```

harbaum, 2023-08-18:
Perhaps not unrelated: I have seen very similar problems with the USB host, although there the problems were when writing to the device. So my assumption is, that this is some kind of generic DMA problem or similar and not a problem with SDH or USBH.

sakumisu, 2023-08-28:
This is a cache issue, and `full test data to buff` is not right also, we will push patch later.

sakumisu, 2023-08-28:
Have done.

harbaum, 2023-08-29:
Indeed, the test now passes. Thanks a lot!

closed 2023-08-29 by harbaum