bouffalolab/bouffalo_sdk #179: BL616 flashing and debugging

JohnnyZig, 2023-10-16:
I am evaluating BL616 (Sipeed M0s) for use in my project.
Currently I am using STM32 MCUs and experience is pretty slipstreamed - I am building my code with LLVM compiler and flashing&debugging with GDB (with BlackMagicProbe which is GDB server implemented inside another STM32 board - no external programs like OpenOCD needed).

Is that possible with BL616 in current state?
Can I use LLVM to building code instead of GCC?
Can I flash my code to the board with (non GUI) opensource tool (blisp doesn't support BL616 yet)?
Can I debug (breakpoint, single stepping, variable/register watch) my code from GDB/LLDB without need for 3rd party programs and configs (like OpenOCD or T-Head debug server)?

gamelaster, 2023-10-16:
You can use LLVM if you use LHAL directly, or configure CMake to use LLVM, but how well it will compile, dunno.

Beside blisp, there is no open source flasher. BL616 Support might arrive around December, but no promises.

About debugging, it is possible with OpenOCD.

JohnnyZig, 2023-10-18:
Is it possible to flash it and restart via JTAG with OpenOCD?

gamelaster, 2023-10-18:
At the moment, nope.

Pavlos1, 2023-11-01:
`bflb-iot-tool` and `bflb-mcu-tool` appear to support the BL616, but I do not have the hardware to test.

They are MIT licensed and available via `pip install`.

closed 2024-02-07 by sakumisue

JohnnyZig, 2024-02-07:
How this is completed? 
I was not able to debug this MCU on multiple PCBs (even created 2 PCBs for this purpose!).

Tried it with:
    2 Sipeed M0s dock with soldered wires for default JTAG (pins 0-3)
    2 my custom PCB which has that pins broken out to pin header

Tried with and without JTAG pullup&pulldown resistors and with and without JTAG in series resistors.
Tried with multiple JTAG HW adapters:
    Sipeed RV-debugger (FTDI2232D)
    BlackMagic probe on STM32F4 (with RISC-V support which works for GD32V)
    GD32V with chinese firmware which is similar to BlackMagic Probe
    CKlink on STM32F1

Tried it with SW:
    OpenOCD RISC-V (on FreeBSD and Linux)
    Debug Server (which is Linux/Windows only binary) v5.16 which is latest version I could find
    directly with BlackMagic Probe

Tried it with multiple examples from Bouffalo's and Sipeed's SDK.

OpenOCD recognizes the CPU core but that's all:
Info : JTAG tap: riscv.cpu tap/device found: 0x10000b6f (mfg: 0x5b7 (T-HEAD Semiconductor Co Ltd), part: 0x0000, ver: 0x1)

Looking at electrical signals with logic analyzer it seems that replying to JTAG IDCODE works, but later reading RISC-V CSR registers (0x11) fails.
Also, OpenOCD will only recognize MCU after fresh power-on-restart of MCU.

I had plans to play with this MCU over Christmas, spent already two months with it and I can't even single step it in debugger (not to mention flash it over JTAG/GDB).
HW is pretty impressive, but it is useless without proper tools and documentation.
Where is clock chapter in datasheet/reference manual?
Where is support in standard opensource tools like upstream OpenOCD, upstream GDB and upstream GCC?
LLVM seems to be working (at least for my simple blinky).

In contrast I did manage to run ESP32 (which I didn't use in my life) for the first time while drinking my morning coffee. They have less stellar HW yes (they do have builtin USB JTAG which works), but they have documentation, examples, community and they don't close unresolved tickets...

gamelaster, 2024-02-07:
@JohnnyZig 
> Where is support in standard opensource tools like upstream OpenOCD, upstream GDB and upstream GCC?

You can use upstream OpenOCD with BL618, it works fine.
GDB and GCC, you need to use T-Head toolchain, which is open-source and maintained by T-Head (this is not Bouffalo's responsibility): https://github.com/T-head-Semi/xuantie-gnu-toolchain

About JTAG, I had issues with Sipeed M0s Dock and other Sipeed BL618 boards. On Sipeed boards, I needed to multiplex JTAG to another pins. Although, I am pretty sure that JTAG & GDB works in combination with Debug Server, as we dumped BootROM this way from BL618.