bouffalolab/bouffalo_sdk #183: BLE Audio

mrhempman69, 2023-11-07:
I was directed here from the examples repo: https://github.com/sipeed/M0S_BL616_example/issues/13

Can you please provide an example of how to set up the device to be used as a BLE Audio acceptor, from a phone connection? And the example can show both how to use I2S on the output, and how to use the DAC on the output too?

Where can i get more information on how to use BLE audio with this chip? I know it will involve using one of the LC3 codec libraries on Github

sakumisu, 2023-11-09:
examples/audio/player_bt bt audio here.

mrhempman69, 2023-11-10:
Hello, I have successfully flashed M0S dock with https://github.com/bouffalolab/bouffalo_sdk/tree/master/examples/audio/player_bt

The problem is, that I cannot execute the A2DP start command, because there is no UART terminal available when plugged in over USB, nor when using the boards GND TX RX pins.

gamelaster, 2023-11-10:
@mrhempman69 there are headers with GND RXD TXD on PCB of M0s Dock.

mrhempman69, 2023-11-10:
But which IOs for this example? 

I already tried the holes labeled "T" and "R"

gamelaster, 2023-11-10:
@mrhempman69 that pins labeled T, R and G are those you need.

mrhempman69, 2023-11-10:
Thanks. I do not know why it isn't working. The baud rate is supposed to be 2000000 right? And it's not supposed to show a UART interface when connected by USB-C, I tried with both silicon l abs CP210x UART bridge (usb), and LTD FT232 Serial Uart (usb). 

I have wired it correctly (T<-->R and R<-->T), tried powering from both USB-C and header-power, 

Here is the output from the flash log? Perhaps it will help:

```
[12:08:24.206] - Serial port is COM8
[12:08:24.206] - ==================================================
[12:08:24.226] - FW get address from partiton file C:\Users\mrhem\bouffalo_sdk_setup\bouffalo_sdk\examples\audio\player_bt\./build/build_out\partition.bin
[12:08:24.227] - Address=0x10000
[12:08:24.229] - Program Start
[12:08:24.229] - ========= eflash loader cmd arguments =========
[12:08:24.240] - serial port is COM8
[12:08:24.240] - chiptype: bl616
[12:08:24.241] - cpu_reset=False
[12:08:24.518] - com speed: 2000000
[12:08:24.519] - ========= Interface is uart =========
[12:08:24.520] - Bootrom load
[12:08:24.523] - ========= get_boot_info =========
[12:08:24.524] - ========= image get bootinfo =========
[12:08:24.809] - tx rx and power off, press the machine!
[12:08:24.809] - cutoff time is 0.05
[12:08:24.872] - power on tx and rx
[12:08:25.862] - reset cnt: 0, reset hold: 0.05, shake hand delay: 0.1
[12:08:25.862] - clean buf
[12:08:25.870] - send sync
[12:08:26.095] - ack is b'4f4b'
[12:08:26.098] - shake hand success
[12:08:26.612] - data read is b'010016060000010027928001536336cf0eb417000f758010'
[12:08:26.613] - ========= ChipID: b40ecf366353 =========
[12:08:26.616] - Get bootinfo time cost(ms): 2091.80908203125
[12:08:26.618] - change bdrate: 2000000
[12:08:26.620] - Clock PLL set
[12:08:26.622] - Set clock time cost(ms): 1.00146484375
[12:08:26.751] - Read mac addr
[12:08:26.752] - MACADDR: b40ecf366353
[12:08:26.752] - flash set para
[12:08:26.754] - get flash pin cfg from bootinfo: 0x02
[12:08:26.757] - set flash cfg: 1014102
[12:08:26.759] - Set flash config
[12:08:26.761] - Set para time cost(ms): 1.12890625
[12:08:26.762] - ========= flash read jedec ID =========
[12:08:26.764] - Read flash jedec ID
[12:08:26.765] - readdata:
[12:08:26.766] - b'c8601600'
[12:08:26.767] - Finished
[12:08:26.772] - flash config Not found,use default
[12:08:26.773] - jedec_id:c86016
[12:08:26.774] - capacity_id:22
[12:08:26.776] - capacity:4.0M
[12:08:26.777] - get flash size: 0x00400000
[12:08:26.778] - Program operation
[12:08:26.781] - Dealing Index 0
[12:08:26.781] - ========= programming C:\Users\mrhem\bouffalo_sdk_setup\bouffalo_sdk\examples\audio\player_bt\.\build\build_out\boot2_bl616_isp_release_v8.1.1.bin to 0x000000
[12:08:26.785] - flash para file: C:\Users\mrhem\bouffalo_sdk_setup\bouffalo_sdk\tools\bflb_tools\bouffalo_flash_cube\chips/bl616/efuse_bootheader/flash_para.bin
[12:08:26.788] - Set flash config
[12:08:26.790] - Set para time cost(ms): 2.6806640625
[12:08:26.790] - ========= flash load =========
[12:08:26.793] - ========= flash erase =========
[12:08:26.794] - Erase flash  from 0x0 to 0xb76f
[12:08:27.028] - Erase time cost(ms): 233.175048828125
[12:08:27.074] - decompress flash load 23368
[12:08:27.101] - Load 2048/23368 {"progress":8}
[12:08:27.116] - Load 4096/23368 {"progress":17}
[12:08:27.133] - Load 6144/23368 {"progress":26}
[12:08:27.142] - Load 8192/23368 {"progress":35}
[12:08:27.158] - Load 10240/23368 {"progress":43}
[12:08:27.174] - Load 12288/23368 {"progress":52}
[12:08:27.191] - Load 14336/23368 {"progress":61}
[... 472 lines cut ...]
[12:08:38.063] - Load 872448/888608 {"progress":98}
[12:08:38.089] - Load 874496/888608 {"progress":98}
[12:08:38.115] - Load 876544/888608 {"progress":98}
[12:08:38.135] - Load 878592/888608 {"progress":98}
[12:08:38.159] - Load 880640/888608 {"progress":99}
[12:08:38.177] - Load 882688/888608 {"progress":99}
[12:08:38.186] - Load 884736/888608 {"progress":99}
[12:08:38.202] - Load 886784/888608 {"progress":99}
[12:08:38.230] - Load 888608/888608 {"progress":100}
[12:08:38.230] - Load 888608/888608 {"progress":100}
[12:08:38.233] - Write check
[12:08:38.238] - Flash load time cost(ms): 7756.065185546875
[12:08:38.238] - Finished
[12:08:38.268] - Sha caled by host: 2084c0b16898a7de5804c60d3d0c5ced2b77e92b7dcd44c9ceace7fa73c57f88
[12:08:38.268] - xip mode Verify
[12:08:39.509] - Read Sha256/1496336
[12:08:39.509] - Flash xip readsha time cost(ms): 1237.662353515625
[12:08:39.514] - Finished
[12:08:39.516] - Sha caled by dev: 2084c0b16898a7de5804c60d3d0c5ced2b77e92b7dcd44c9ceace7fa73c57f88
[12:08:39.517] - Verify success
[12:08:39.522] - Program Finished
[12:08:39.523] - All time cost(ms): 15293.658203125
[12:08:39.658] - close interface
[12:08:39.658] - [All Success]
```

mrhempman69, 2023-11-10:
Is there a line that I can change in the code to aytomatically enable a2dp broadcast? Without needing uart.

Also the Bluetooth name if possible. So that i can flash it with those settings and not use UART

mrhempman69, 2023-11-11:
This is a problem specific to the `player_bt` example. Because, I can run `helloworld` example and get proper UART output.

But, even when I make the following changes to `examples/audio/player_bt` (and add LOG statements similar to helloworld), still not outputting over UART connection. No function:

```
#include "shell.h"
#include <FreeRTOS.h>
#include "task.h"
#include "board.h"

#define APP_MAIN_STACK_SIZE          (4*1024)
#define APP_MAIN_TASK_PRIORITY       (1)

static struct bflb_device_s *uart0;

extern void shell_init_with_task(struct bflb_device_s *shell);
extern void app_main_entry(void *arg);

int main(void)
{
    TaskHandle_t xHandle;
    board_init();

    // configASSERT((configMAX_PRIORITIES > 4));

    uart0 = bflb_device_get_by_name("uart0");
    shell_init_with_task(uart0);

    xTaskCreate(app_main_entry, (char *)"app_main", APP_MAIN_STACK_SIZE, NULL, APP_MAIN_TASK_PRIORITY, &xHandle);

    vTaskStartScheduler();

    **while (1) {
        LOG_F("player_bt hello world fatal\r\n");
        LOG_E("player_bt hello world error\r\n");
        LOG_W("player_bt hello world warning\r\n");
        LOG_I("player_bt hello world information\r\n");
        LOG_D("player_bt hello world debug\r\n");
        LOG_T("player_bt hello world trace\r\n");
        LOG_RF("player_bt hello world fatal raw\r\n");
        LOG_RE("player_bt hello world error raw\r\n");
        LOG_RW("player_bt hello world warning raw\r\n");
        LOG_RI("player_bt hello world information raw\r\n");
        LOG_RD("player_bt hello world debug raw\r\n");
        LOG_RT("player_bt hello world trace raw\r\n");
        bflb_mtimer_delay_ms(3000);
    }**
}
```

Please try audio/player_bt example yourself and see if you get UART output.

mrhempman69, 2023-11-11:
So the `helloworld` example works. But the `examples/audio/player_bt` one does not. Even when I change its `main.c` to be more like helloworld, still no logging output. 

I think it has to do with `bflb_device_get_by_name` function...

```
/*
 * Copyright (C) 2017-2022 Bouffalolab Group Holding Limited
 * Change Logs:
 *   Date        Author       Notes
 *   2019-03-25  MSP          the first version
 */

#include "shell.h"
#include <FreeRTOS.h>
#include "task.h"
#include "bflb_mtimer.h"
#include "bflb_core.h"
#include "board.h"
#define DBG_TAG "MAIN"
#include "log.h"

#define APP_MAIN_STACK_SIZE          (4*1024)
#define APP_MAIN_TASK_PRIORITY       (1)

static struct bflb_device_s *uart0;

extern void shell_init_with_task(struct bflb_device_s *shell);
extern void app_main_entry(void *arg);

int main(void)
{
    TaskHandle_t xHandle;
    board_init();

    // configASSERT((configMAX_PRIORITIES > 4));

    uart0 = bflb_device_get_by_name("uart0");
    shell_init_with_task(uart0);

    xTaskCreate(app_main_entry, (char *)"app_main", APP_MAIN_STACK_SIZE, NULL, APP_MAIN_TASK_PRIORITY, &xHandle);

    vTaskStartScheduler();

   // No output in this example, but works in helloworld:
    while (1) {
        LOG_F("player_bt hello world fatal\r\n");
        LOG_E("player_bt hello world error\r\n");
        LOG_W("player_bt hello world warning\r\n");
        LOG_I("player_bt hello world information\r\n");
        LOG_D("player_bt hello world debug\r\n");
        LOG_T("player_bt hello world trace\r\n");
        LOG_RF("player_bt hello world fatal raw\r\n");
        LOG_RE("player_bt hello world error raw\r\n");
        LOG_RW("player_bt hello world warning raw\r\n");
        LOG_RI("player_bt hello world information raw\r\n");
        LOG_RD("player_bt hello world debug raw\r\n");
        LOG_RT("player_bt hello world trace raw\r\n");
        bflb_mtimer_delay_ms(3000);
    }
}

```

sakumisu, 2023-11-12:
M0s have no psram, stop using these demos.

gamelaster, 2023-11-12:
@mrhempman69 as @sakumisu says, those examples needs pSRAM, so they can work properly. I can recommend AI-Thinker M61 series boards: https://www.aliexpress.com/i/1005005407935386.html

mrhempman69, 2023-11-12:
Thanks... it has to be "p"SRAM? Would like this chip from Adafruit would work? https://www.digikey.com/en/products/filter/memory/memory/774?s=N4IgjCBcoKwOwAYqgMZQGYEMA2BnApgDQgD2UA2iAMwCcCAbAEwAsIAusQA4AuUIAytwBOASwB2AcxABfYmEY0I0EGkhY8RUhRALmCBSGK64MGIZ01mNRvXO6aMRuZgJEADlZHLNOLa8uYNzsHBk8LF3ozDhAePkFRSRliAFonZVVhAFdNMkhKKOlZEFtlEQATPmSwBAguXkgQcwBHbgBPPmrzNs58PkxcNEKgA

8 x 8MB memory
2.7-3.6 operating voltage
8-SOIC

It's $1.75 though, while SRAM chips have more options and are cheaper.... 
SPI interface 
133 MHz clock 

133 MHz
--

<br class="Apple-interchange-newline">

gamelaster, 2023-11-12:
No, this will not work. You need the pSRAM integrated directly in the chip, afaik, BL616 doesn't expose controller on GPIOs for external pSRAM usage.

sakumisu, 2023-11-12:
psram is chip built-in, you should use bl618

mrhempman69, 2023-11-12:
The Bl618 doesn’t come in small modules like this: https://www.aliexpress.us/item/3256805074088612.html?_randl_currency=USD&src=google&src=google&albch=shopping&acnt=708-803-3821&slnk=&plac=&mtctp=&albbt=Google_7_shopping&albagn=888888&isSmbAutoCall=false&needSmbHouyi=false&albcp=20268592310&albag=&trgt=&crea=en3256805074088612&netw=x&device=m&albpg=&albpd=en3256805074088612&gbraid=0AAAAAD6I-hFFEFLRuckLvKdIvdb82cY6A&gclid=Cj0KCQiAjMKqBhCgARIsAPDgWlxiNlrkhrMFzi3KhlxweuxT_1HmwylXg_2fn7QtvjHduc8pWik32jYaAguUEALw_wcB&gclsrc=aw.ds&aff_fcid=2259b38bd9cf487092381962c687a7f0-1699798326242-09110-UneMJZVf&aff_fsk=UneMJZVf&aff_platform=aaf&sk=UneMJZVf&aff_trace_key=2259b38bd9cf487092381962c687a7f0-1699798326242-09110-UneMJZVf&terminal_id=e667f575eb6d4a36b049600e6e592c43&afSmartRedirect=y&gatewayAdapt=glo2usa&_randl_shipto=US

Where do I buy the bl618 chip only (workable breakout, not QFN, not DIP, size in between)

Why does the README  for the player_bt example specifically mention BL616 so many times if it doesn’t work for BL616?

It’s totally unsupported to get BLE audio to work for the more common BL616?

madushan1000, 2023-11-13:
There is a bl618 module from Ai-Thinker(AI-M61-32S). Also some shops on aliexpress sells bl618 chips too. just search for "bl618"

mrhempman69, 2023-11-13:
Thank you, madushan1000, gamelaster !!!

mrhempman69, 2023-11-24:
Okay??? I bought the AI-Thinker-M61 board that was suggested. I get this error when connecting to UART after flash.

```
[15:00:22.162] - Open COM4 Success
[15:00:23.486] -   ____               __  __      _       _       _
[15:00:23.494] -  |  _ \             / _|/ _|    | |     | |     | |
[15:00:23.501] -  | |_) | ___  _   _| |_| |_ __ _| | ___ | | __ _| |__
[15:00:23.505] -  |  _ < / _ \| | | |  _|  _/ _` | |/ _ \| |/ _` | '_ \
[15:00:23.509] -  | |_) | (_) | |_| | | | || (_| | | (_) | | (_| | |_) |
[15:00:23.513] -  |____/ \___/ \__,_|_| |_| \__,_|_|\___/|_|\__,_|_.__/
[15:00:23.515] - Build:11:39:57,Nov 10 2023
[15:00:23.518] - Copyright (c) 2022 Bouffalolab team
[15:00:23.519] - ======== flash cfg ========
[15:00:23.522] - flash size 0x00800000
[15:00:23.523] - jedec id     0x5E4017
[15:00:23.525] - mid              0x5E
[15:00:23.526] - iomode           0x04
[15:00:23.529] - clk delay        0x01
[15:00:23.530] - clk invert       0x01
[15:00:23.532] - read reg cmd0    0x05
[15:00:23.533] - read reg cmd1    0x35
[15:00:23.535] - write reg cmd0   0x01
[15:00:23.535] - write reg cmd1   0x31
[15:00:23.536] - qe write len     0x01
[15:00:23.538] - cread support    0x01
[15:00:23.539] - cread code       0x20
[15:00:23.541] - burst wrap cmd   0x77
[15:00:23.542] - ===========================
[15:00:23.547] - dynamic memory init success, ocram heap size = 234 Kbyte, psram heap size = 4096 Kbyte
[15:00:23.547] - sig1:ffffffff
[15:00:23.549] - sig2:0000f32f
[15:00:23.550] - cgen1:9ffffffd
[15:00:23.552] - [0mbouffalolab />[0mexception_entry
[15:00:23.553] - mcause=38000005
[15:00:23.553] - mepc:a00d9846
[15:00:23.555] - mtval:00000001
[15:00:23.556] - Load access fault
```

I am using BL618 now, still the repo is broken. Still the repo readme suggests using BL616, why don't you update it? This conversation was two weeks ago, still nobody has updated the documentation.

sakumisu, 2023-11-29:
These demos can work both bl616 or bl618, just modify some config, but i should say these demos are for customer with special board. From mine, it work. You can ask ai-thinker for supporting these, thanks.

mrhempman69, 2023-11-29:
I made a Bl618 breakout board too now.  So I can re-create "special board" but what I should do to make it like the "special board"? What is the special board? Where is its schematic?

 And also where are the config settings are that I need to change, and for what reasons?

Need to document this stuff if you want you and the company to make money

gamelaster, 2023-11-29:
This is correct command. @sakumisu uses probably the official devkit. Best would be to attach JTAG and check out why it crashes.

mrhempman69, 2023-11-30:
@gamelaster  Can you please recommend me the CKLink Debugger or the jtag tool that you're using to debug these kinds of boards?

gamelaster, 2023-12-01:
Sipeed BL616 board is supposed to be CK-Link debugger :grimacing:

mrhempman69, 2023-12-01:
You mean this M0S dock? How do you flash and wire that as a debugger? 
[image]

O2C14, 2024-02-25:
> Okay??? I bought the AI-Thinker-M61 board that was suggested. I get this error when connecting to UART after flash of the player_bt example. (Has anyone actually tried the player_bt example??)
> 
> ```
> [15:00:22.162] - Open COM4 Success
> [15:00:23.486] -   ____               __  __      _       _       _
> [15:00:23.494] -  |  _ \             / _|/ _|    | |     | |     | |
> [15:00:23.501] -  | |_) | ___  _   _| |_| |_ __ _| | ___ | | __ _| |__
> [15:00:23.505] -  |  _ < / _ \| | | |  _|  _/ _` | |/ _ \| |/ _` | '_ \
> [15:00:23.509] -  | |_) | (_) | |_| | | | || (_| | | (_) | | (_| | |_) |
> [15:00:23.513] -  |____/ \___/ \__,_|_| |_| \__,_|_|\___/|_|\__,_|_.__/
> [15:00:23.515] - Build:11:39:57,Nov 10 2023
> [15:00:23.518] - Copyright (c) 2022 Bouffalolab team
> [15:00:23.519] - ======== flash cfg ========
> [15:00:23.522] - flash size 0x00800000
> [15:00:23.523] - jedec id     0x5E4017
> [15:00:23.525] - mid              0x5E
> [15:00:23.526] - iomode           0x04
> [15:00:23.529] - clk delay        0x01
> [15:00:23.530] - clk invert       0x01
> [15:00:23.532] - read reg cmd0    0x05
> [15:00:23.533] - read reg cmd1    0x35
> [15:00:23.535] - write reg cmd0   0x01
> [15:00:23.535] - write reg cmd1   0x31
> [15:00:23.536] - qe write len     0x01
> [15:00:23.538] - cread support    0x01
> [15:00:23.539] - cread code       0x20
> [15:00:23.541] - burst wrap cmd   0x77
> [15:00:23.542] - ===========================
> [15:00:23.547] - dynamic memory init success, ocram heap size = 234 Kbyte, psram heap size = 4096 Kbyte
> [15:00:23.547] - sig1:ffffffff
> [15:00:23.549] - sig2:0000f32f
> [15:00:23.550] - cgen1:9ffffffd
> [15:00:23.552] - �[0mbouffalolab />�[0mexception_entry
> [15:00:23.553] - mcause=38000005
> [15:00:23.553] - mepc:a00d9846
> [15:00:23.555] - mtval:00000001
> [15:00:23.556] - Load access fault
> ```
> 
> I am using BL618 now, still the repo is broken. Still the repo readme suggests using BL616, why don't you update it? This conversation was two weeks ago, still nobody has updated the documentation.

The occurrence of this phenomenon is due to compiling in the Windows environment. Please try compiling in the Linux environment

Virus-V, 2026-05-13:
Closing as not planned — the requested BLE Audio (LE Audio with LC3) example does not yet exist in the SDK.

**Current state:**
- `examples/audio/player_bt` provides **Classic Bluetooth A2DP** (SBC codec), not BLE Audio/LE Audio
- The LC3 codec library is integrated in `components/wireless/bluetooth/btblestack/src/lc3/`
- LE Audio API headers (`bt_audio_unicast_group`, `bt_audio_broadcast_sink/source`) exist in the btblestack
- CIS/BIS (ISO) support is present in the controller but not enabled in default configs
- A full LE Audio sink/acceptor example has not been implemented

For classic BT audio (A2DP sink), please refer to `examples/audio/player_bt` or `examples/btble/a2dp_player`. Note that these examples require a chip with built-in PSRAM (BL618).