bouffalolab/bouffalo_sdk #187: Please allolw USB host and device at the same time

harbaum, 2023-11-28:
Please change this:

https://github.com/bouffalolab/bouffalo_sdk/blob/81a1645805de9ebb83666946d1daebbe8fcd7025/components/usb/cherryusb/CMakeLists.txt#L45

to

```
endif()
if(CONFIG_CHERRYUSB_HOST)
```

This will allow to use host and device at the same time.

My use case is a device that usually acts as a USB host. But it can optionally be connected to a PC for update. And in that case it needs to be a USB device.

I have tested this and with this change the device can switch roles and works as expected.

gamelaster, 2023-11-28:
@harbaum I think it would be better to ask about this here: https://github.com/cherry-embedded/CherryUSB

harbaum, 2023-11-28:
No. This CMakefile only exists in the bouffalo sdk and not in cherryusb.

Virus-V, 2025-12-29:
hardware not support