bouffalolab/bouffalo_sdk #197: IDE setup

nimish, 2024-01-31:
The cmake/makefile mix of the repo makes it confusing to set up in an IDE. Is there a guide?

Jatsekku, 2024-02-04:
Yeah, I agree, 
@sakumisue  could you provide example how to use SDK as external dependency and how to compile out-of-tree application code on top of SDK?

sakumisu, 2024-02-07:
@Jatsekku Just copy helloworld out of sdk and modify BL_SDK_BASE will work.

sakumisu, 2024-02-07:
> The cmake/makefile mix of the repo makes it confusing to set up in an IDE. Is there a guide?
> 
> Alternatively I stripped out the custom CMake code here: https://github.com/nimish/bl616-tmpl but it's not as complete

Sorry sir, i need to say your project has many problems.
- When cmake builds done, your variables can not be modified, like COMX,
- When we use other chips, we need to modify cmake file, no, we don't allow that
- We don't allow these codes, we should add automatically

```
target_link_libraries(lhal PUBLIC
    soc
    phyrf
    cherryusb)

```
    )
- and many other problems i do not want to say, so i suggest you remove your project and do not have such idea.

sakumisu, 2024-02-07:
Makefile is just to run cmake cmd and modifiy some variables.

sakumisu, 2024-02-07:
Like this, if you can write cmake project, i think you can know these comand, so where is mix?

```
build:Makefile
	$(CMAKE) -S . -B build -G $(cmake_generator) $(cmake_definition)
	$(MAKE) -C build -j8
	$(MAKE) -C build combine

```

nimish, 2024-02-07:
> > The cmake/makefile mix of the repo makes it confusing to set up in an IDE. Is there a guide?
> > Alternatively I stripped out the custom CMake code here: https://github.com/nimish/bl616-tmpl but it's not as complete
> 
> Sorry sir, i need to say your project has many problems.
> 
> * When cmake builds done, your variables can not be modified, like COMX,
> * When we use other chips, we need to modify cmake file, no, we don't allow that
> * We don't allow these codes, we should add automatically
> 
> ```
> target_link_libraries(lhal PUBLIC
>     soc
>     phyrf
>     cherryusb)
> ```
> 
>  * If we use cmake to build, it will be too much config to set, we don't allow that
> * and many other problems i do not want to say, so i suggest you remove your project and do not have such idea.

In general, I would like to know if there is a way to get a template project set up in CLion or VSCode using the existing makefiles and CMake. 

The bl616-tmpl was my personal reorganization for the hw I use that integrates with the IDE and is simply an example. It supports incremental and parallel building along with IDE cmake integration so it is faster to iterate. The price of that is the explicit dependency handling. It is not something that the SDK itself should adopt without significant extra development to support other chips.

gamelaster, 2024-02-07:
I am using Bouffalo SDK with my project through CLion, and it works perfectly. 
This is how I have it configured:
CMake Options: `-G "Unix Makefiles" -DCROSS_COMPILE=/home/gamiee/Desktop/tools/toolchain_gcc_t-head_linux/bin/riscv64-unknown-elf- -DCHIP=bl616 -DCPU_ID= -DBOARD=bl616dk -DBOARD_DIR= -DCONFIG_DEBUG=y -DCONFIG_ROMAPI=y -DCONFIG_USB_HS=y -DCONFIG_COMX=COM5 -DCMAKE_EXPORT_COMPILE_COMMANDS=OFF -DCONFIG_TLSF=y`
Environment: `BL_SDK_BASE=/home/gamiee/Desktop/tools/bouffalo_sdk`

My CMakeLists.txt:
[image]

I was even able to make auto-upload by Run command with terminal. If there will be interest, I can create sample CMake project with CLion configuration stored here.

sakumisu, 2024-02-07:
> I am using Bouffalo SDK with my project through CLion, and it works perfectly. This is how I have it configured: (NOTE: There are no Bouffalo SDK changes) CMake Options: `-G "Unix Makefiles" -DCROSS_COMPILE=/home/gamiee/Desktop/tools/toolchain_gcc_t-head_linux/bin/riscv64-unknown-elf- -DCHIP=bl616 -DCPU_ID= -DBOARD=bl616dk -DBOARD_DIR= -DCONFIG_DEBUG=y -DCONFIG_ROMAPI=y -DCONFIG_USB_HS=y -DCONFIG_COMX=COM5 -DCMAKE_EXPORT_COMPILE_COMMANDS=OFF -DCONFIG_TLSF=y` Environment: `BL_SDK_BASE=/home/gamiee/Desktop/tools/bouffalo_sdk`
> 
> My CMakeLists.txt: ![image](https://private-user-images.githubusercontent.com/1665373/302914940-33d209ff-5661-49fa-9702-81d29cf329fc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDcyOTcwMTEsIm5iZiI6MTcwNzI5NjcxMSwicGF0aCI6Ii8xNjY1MzczLzMwMjkxNDk0MC0zM2QyMDlmZi01NjYxLTQ5ZmEtOTcwMi04MWQyOWNmMzI5ZmMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIwNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMDdUMDkwNTExWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MjZjYjI2NzA2MDk5ODk3ZDUxOGNhOGE5ZGRmMWRmYzBkYWQ3OTJiYjJlYTI2YzNjOWYzMzE2ZGM1MTk0ZWMyYyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.KmG5En5iD37wOq8Z40k5NxHgupU61HhDjK7Ywvo05rI)
> 
> As @sakumisu mentioned, there is no need for using Make files, everything can be easily used through CMake, with CLion or VS Code CMake integration.
> 
> I was even able to make auto-upload by Run command with terminal. If there will be interest, I can create sample CMake project with CLion configuration stored here.

Right, copy cmake comand from makefile into clion, so makefile is unuseful, just make command length lesss for user.

sakumisu, 2024-02-07:
https://github.com/bouffalolab/bouffalo_sdk/blob/be623ceb7f18c70f3aafba2a0eceba5efc6a683a/project.build#L78

nimish, 2024-02-07:
Great, thanks. I think this covers what I wanted. Perhaps this could be added to the README as well?

closed 2024-02-07 by nimish