bouffalolab/bouffalo_sdk #199: Do not overwrite CMake project() macro

Jatsekku, 2024-02-06:
Due to fact that project() is being redefined at [extension.cmake](https://github.com/bouffalolab/bouffalo_sdk/blob/be623ceb7f18c70f3aafba2a0eceba5efc6a683a/cmake/extension.cmake#L226) after including the boufallo_sdk vi find_package, all CMake's project's are affected and can't be build properly. As far as I know it's not good practice and make it very hard to use sdk as external dependency for other projects.

sakumisu, 2024-02-07:
It is a popular style to redefine project, many projects follow this, and we do not suggest you to use other cmake style.

Jatsekku, 2024-02-07:
So how can it  be used to avoid problem which I mentioned before (without changes in SDK itself)?

sakumisu, 2024-02-08:
Do not use your project's cmakelist, and add a new cmakelist with bl style in your project.

closed 2024-02-09 by Jatsekku