bouffalolab/bouffalo_sdk #202: Enabling Threads - Linux OS

IronFist16, 2024-03-23:
Hello All,

I was trying to test threading with this SDK, looks like I'm not able to wrap my head over the proper library and setup in the CMakeLists.txt file. Can anyone help me please? 

The below is an example of my CMakeLists.txt file:

```
cmake_minimum_required(VERSION 3.15)

include(proj.conf)

find_package(bouffalo_sdk REQUIRED HINTS $ENV{BL_SDK_BASE})

set(CMAKE_THREAD_PREFER_PTHREAD TRUE)
set(THREADS_PREFER_PTHREAD_FLAG TRUE)

find_package(Threads REQUIRED)

if(Threads_FOUND)
    message(STATUS "Thread Library found ${CMAKE_THREAD_LIBS_INIT}")
    if(CMAKE_USE_PTHREADS_INIT
    target_compile_options(app PRIVATE -pthread)
    target_link_options(app PRIVATE -pthread)
    message(STATUS "Pthread included !")
    else()
    message(WARNING "pthread not included !")
    endif()
endif()

target_link_libraries(app ${CMAKE_THREAD_LIBS_INIT})

sdk_set_main_file(main.c)

project(helloworld)
```

 while I do `#include <pthread.h>` , I can see the below **warning** only message during compilation, which end successfully:

```
/usr/bin/toolchain_gcc_t-head_linux/bin/../lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: CMakeFiles/helloworld_bl808_m0.elf.dir/main.c.o: in function `main':
/home/ironfist/Documents/Robot/Docs/M1S_Dock/bl808_linux/bouffalo_sdk/examples/helloworld/main.c:117: warning: pthread_create is not implemented and will always fail
/usr/bin/toolchain_gcc_t-head_linux/bin/../lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /home/ironfist/Documents/Robot/Docs/M1S_Dock/bl808_linux/bouffalo_sdk/examples/helloworld/main.c:122: warning: pthread_join is not implemented and will always fail
```
Below is the complete log.

```

./../../tools/cmake/bin/cmake -S . -B build -G "Unix Makefiles" -DCROSS_COMPILE=riscv64-unknown-elf- -DCHIP=bl808 -DCPU_ID=m0 -DBOARD=bl808dk -DBOARD_DIR= -DCONFIG_DEBUG=y -DCONFIG_ROMAPI=y -DCONFIG_USB_HS=y -DCONFIG_COMX=COM5 -DCMAKE_EXPORT_COMPILE_COMMANDS=OFF -DCONFIG_TLSF=y -DCONFIG_COREDUMP=y
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/toolchain_gcc_t-head_linux/bin/riscv64-unknown-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/toolchain_gcc_t-head_linux/bin/riscv64-unknown-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/toolchain_gcc_t-head_linux/bin/riscv64-unknown-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- [register library : libc], path:/home/ironfist/Documents/Robot/Docs/M1S_Dock/bl808_linux/bouffalo_sdk/components/libc
-- [register library : mm], path:/home/ironfist/Documents/Robot/Docs/M1S_Dock/bl808_linux/bouffalo_sdk/components/mm
-- [register library : utils], path:/home/ironfist/Documents/Robot/Docs/M1S_Dock/bl808_linux/bouffalo_sdk/components/utils
-- [register library : lhal], path:/home/ironfist/Documents/Robot/Docs/M1S_Dock/bl808_linux/bouffalo_sdk/drivers/lhal
-- [register extern library : libcsi_xt900p32f_dsp], path:/home/ironfist/Documents/Robot/Docs/M1S_Dock/bl808_linux/bouffalo_sdk/drivers/lhal
-- [register extern library : libpka], path:/home/ironfist/Documents/Robot/Docs/M1S_Dock/bl808_linux/bouffalo_sdk/drivers/lhal
-- [register library : std], path:/home/ironfist/Documents/Robot/Docs/M1S_Dock/bl808_linux/bouffalo_sdk/drivers/soc/bl808/std
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
[... 189 lines cut ...]
[03:03:53.737] - Image Offset:0x2000
[03:03:53.737] - Image hash is b'e654721f94cf21a36d9a506c406e87ad58902147378cc69ee8af56978ab4aa5c'
[03:03:53.738] - Encrypt efuse data
[03:03:53.739] - Image Offset:0x2000
[03:03:53.739] - Image hash ignore,not calculate
[03:03:53.739] - Bootheader config crc: b'ffe76ec5'
[03:03:53.739] - 
Process /home/ironfist/Documents/Robot/Docs/M1S_Dock/bl808_linux/bouffalo_sdk/examples/helloworld/build/build_out/mfg_bl808_gu_8476f7743_v0.59.bin
[03:03:53.740] - ========= sp image create =========
[03:03:53.740] - Flash config crc: b'4fb1fe70'
[03:03:53.741] - Clock config crc: b'0a894b86'
[03:03:53.741] - Flash  Image
[03:03:53.741] - Image Offset:0x1000
[03:03:53.743] - Image hash is b'fab0b31dd27d4dabde2a644292c34a6566921af6429b5149629e29b5df88f151'
[03:03:53.744] - Encrypt efuse data
[03:03:53.746] - Image Offset:0x1000
[03:03:53.746] - Image hash ignore,not calculate
[03:03:53.746] - Bootheader config crc: b'e172682a'
[03:03:53.747] - create OTA file:/home/ironfist/Documents/Robot/Docs/M1S_Dock/bl808_linux/bouffalo_sdk/examples/helloworld/build/build_out/helloworld_bl808_m0.bin.ota
[03:03:53.768] - create XZ file:/home/ironfist/Documents/Robot/Docs/M1S_Dock/bl808_linux/bouffalo_sdk/examples/helloworld/build/build_out/helloworld_bl808_m0.xz
[03:03:53.804] - create XZ OTA file:/home/ironfist/Documents/Robot/Docs/M1S_Dock/bl808_linux/bouffalo_sdk/examples/helloworld/build/build_out/helloworld_bl808_m0.xz.ota
Built target combine
ironfist@Iron-Fist-Box:~/Documents/Robot/Docs/M1S_Dock/bl808_linux/bouffalo_sdk/examples/helloworld$ 

```

IronFist16, 2024-03-24:
Just to add, I though the compiler had the thread flag disabled, so I checked the version details,  where I can see that:

**`enable-threads=posix`**
**`Thread model: posix`**
```

ironfist@Iron-Fist-Box:~/Documents/Robot/Docs/M1S_Dock/bl808_linux$ riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/usr/bin/toolchain_gcc_t-head_linux/bin/../libexec/gcc/riscv64-unknown-elf/10.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /mnt/ssd/jenkins_iotsw/slave/workspace/Toolchain/build-gnu-riscv/./source/riscv/riscv-gcc/configure --target=riscv64-unknown-elf --with-gmp=/mnt/ssd/jenkins_iotsw/slave/workspace/Toolchain/build-gnu-riscv/build-gcc-riscv64-unknown-elf/build-Xuantie-900-gcc-elf-newlib-x86_64-V2.6.1/lib-for-gcc-x86_64-linux --with-mpfr=/mnt/ssd/jenkins_iotsw/slave/workspace/Toolchain/build-gnu-riscv/build-gcc-riscv64-unknown-elf/build-Xuantie-900-gcc-elf-newlib-x86_64-V2.6.1/lib-for-gcc-x86_64-linux --with-mpc=/mnt/ssd/jenkins_iotsw/slave/workspace/Toolchain/build-gnu-riscv/build-gcc-riscv64-unknown-elf/build-Xuantie-900-gcc-elf-newlib-x86_64-V2.6.1/lib-for-gcc-x86_64-linux --with-libexpat-prefix=/mnt/ssd/jenkins_iotsw/slave/workspace/Toolchain/build-gnu-riscv/build-gcc-riscv64-unknown-elf/build-Xuantie-900-gcc-elf-newlib-x86_64-V2.6.1/lib-for-gcc-x86_64-linux --with-libmpfr-prefix=/mnt/ssd/jenkins_iotsw/slave/workspace/Toolchain/build-gnu-riscv/build-gcc-riscv64-unknown-elf/build-Xuantie-900-gcc-elf-newlib-x86_64-V2.6.1/lib-for-gcc-x86_64-linux --with-pkgversion='Xuantie-900 elf newlib gcc Toolchain V2.6.1 B-20220906' CXXFLAGS='-g -O2 -DTHEAD_VERSION_NUMBER=2.6.1 ' --enable-libgcctf --prefix=/mnt/ssd/jenkins_iotsw/slave/workspace/Toolchain/build-gnu-riscv/build-gcc-riscv64-unknown-elf/Xuantie-900-gcc-elf-newlib-x86_64-V2.6.1 --disable-shared --**enable-threads=posix** --enable-languages=c,c++ --with-system-zlib --enable-tls --with-newlib --with-sysroot=/mnt/ssd/jenkins_iotsw/slave/workspace/Toolchain/build-gnu-riscv/build-gcc-riscv64-unknown-elf/Xuantie-900-gcc-elf-newlib-x86_64-V2.6.1/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=/mnt/ssd/jenkins_iotsw/slave/workspace/Toolchain/build-gnu-riscv/./source/riscv/riscv-gcc --enable-multilib --with-abi=lp64d --with-arch=rv64gcxthead 'CFLAGS_FOR_TARGET=-Os   -mcmodel=medany' 'CXXFLAGS_FOR_TARGET=-Os   -mcmodel=medany'
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 10.2.0 (Xuantie-900 elf newlib gcc Toolchain V2.6.1 B-20220906) 
```

gamelaster, 2024-03-25:
You can't use POSIX threads in Bouffalo SDK, as it does not provide POSIX compatible threading. That's mainly because this SDK is without any RTOS by default, and optionally you can use FreeRTOS, but you still need to use their functions/API, instead of classic pthread headers.

IronFist16, 2024-03-25:
Yes I get that, 

However, in the SDK the, the Linux kernel is there, so upon checking elsewhere I understood for that to work I will need to compile the code with [Linux tool chain](https://gitee.com/bouffalolab/linuxtoolchain_gcc_t-head) instead.

 1-Is this correct? 

Eventually, I downloaded the toolchain and setup the env. var then upon compilation, I face another issues:

```
/home/ironfist/Documents/Robot/Docs/M1S_Dock/bl808_linux/bouffalo_sdk/components/libc/newlib/port_memory.c:2:10: fatal error: reent.h: No such file or directory
    2 | #include <reent.h>
      |          ^~~~~~~~~

/home/ironfist/Documents/Robot/Docs/M1S_Dock/bl808_linux/bouffalo_sdk/components/libc/./sys/types.h:20:10: fatal error: _ansi.h: No such file or directory
   20 | #include <_ansi.h>
      |          ^~~~~~~~~
```

Which I assume it's due to the difference between the **elf** (Newlib) & **linux** (libc) tool chain.

2 - Is this also correct?

Probably I missed updating configuration in the make files or compilation flags, 

3-Can you please explain to me how to properly migrate into the linuxtool chain in this case?