bouffalolab/bouffalo_sdk #206: usbhost example won't compile due to cdc_acm / cdc_ecm mixture

harbaum, 2024-04-22:
The cmakefiles refer to a file named `class/cdc/usbh_cdc_ecm.c`

https://github.com/bouffalolab/bouffalo_sdk/blob/0f07c3b8ecbed6b5eda90cc794611c6dd9077851/components/usb/cherryusb/CMakeLists.txt#L59

but in the repository only a file named` class/cdc/usbh_cdc_acm.c` is actually included:

https://github.com/bouffalolab/bouffalo_sdk/tree/master/components/usb/cherryusb/class/cdc

This prevents the usbhost demo from being compilable.

harbaum, 2024-04-22:
Perhaps ECM should just be disabled in the project config:

https://github.com/bouffalolab/bouffalo_sdk/blob/0f07c3b8ecbed6b5eda90cc794611c6dd9077851/examples/peripherals/usbhost/proj.conf#L4

It's not enabled by default in the demo itself, anyway:
https://github.com/bouffalolab/bouffalo_sdk/blob/0f07c3b8ecbed6b5eda90cc794611c6dd9077851/examples/peripherals/usbhost/usb_host.c#L12

WGR7, 2024-05-08:
Hi,

Do you got the new release version of CherryUSB running on this SDK?

harbaum, 2024-05-08:
Yes. But I only tested the host part.

WGR7, 2024-05-08:
> Yes. But I only tested the host part.

I've ported the CherryUSB lib, but seems that interrupt isn't happening. Could you share your port?
I'm facing a strange problem with my composite device(MSC+CDC). Sporadically, in the middle of transfer, the device is enumerated again but without a CPU reset. I will update the lib to test and open a case, but is really a mess because the problem should be with BL616 port. And the Boufallo support is worse and don't care about a single customer issue. Here on GitHub, there's no answers to the majority of issues opened. The SDK has weeks and months without an update. The documents are missing description about USB peripheral.

BL616 is a great device, but the more important thing is a good ecosystem and SDK are missing. I'm really excited to the esp32-p4 release.

Do you get to debug BL616 through JTAG interface?

harbaum, 2024-05-08:
See https://github.com/harbaum/bouffalo_sdk/

This repo includes an updated cherry USB as a submodule.

Yes, the chip is cool and has a lot of potential and the SDK support is rather disappointing ... 

I have not tried to use jtag.