bouffalolab/bouffalo_sdk #216: Need MQTT +TLS DEMO

staunchheart, 2024-12-11:
need  MQTT +mbedtls demo
components/net/lib/mqtt/MQTT-C/examples/mbedtls_publisher.c
can not work.

gamelaster, 2024-12-11:
what do you mean with "can not work" ?

staunchheart, 2024-12-13:
1、mbedtls_net_connect->     getaddrinfo( host, port, &hints, &addr_list )   return MBEDTLS_ERR_NET_UNKNOWN_HOST
2、change to netconn_gethostbyname(host, &addr); 
           build ok,
           mbedtls_ssl_handshake no error
          mqtt_init   no error
          mqtt_connect no error
          but  server not real connected;
          mqtt_sync/mqtt_publish/mqtt_sync   return MQTT_ERROR_UNKNOWN

staunchheart, 2024-12-13:
> 1、mbedtls_net_connect-> getaddrinfo( host, port, &hints, &addr_list ) return MBEDTLS_ERR_NET_UNKNOWN_HOST
> 2、change to netconn_gethostbyname(host, &addr);
> build ok,
> mbedtls_ssl_handshake no error [...]

staunchheart, 2024-12-13:
> what do you mean with "can not work" ?

1、mbedtls_net_connect-> getaddrinfo( host, port, &hints, &addr_list ) return MBEDTLS_ERR_NET_UNKNOWN_HOST
2、change to netconn_gethostbyname(host, &addr);
build ok,
mbedtls_ssl_handshake no error
mqtt_init no error
mqtt_connect no error
but server not real connected;
mqtt_sync/mqtt_publish/mqtt_sync return MQTT_ERROR_UNKNOWN

gamelaster, 2024-12-13:
Which chip do you use? BL616?

staunchheart, 2024-12-13:
> Which chip do you use? BL616?
yes BL616

staunchheart, 2024-12-13:
can  add a examples to /wifi/sta/xxxx ?

staunchheart, 2024-12-16:
> Which chip do you use? BL616?

can add a examples to /wifi/sta/xxxx ?

gamelaster, 2024-12-17:
sorry, I do not have the example.

staunchheart, 2024-12-18:
> sorry, I do not have the example.

1、mbedtls_net_connect-> getaddrinfo( host, port, &hints, &addr_list ) return MBEDTLS_ERR_NET_UNKNOWN_HOST
2、change to netconn_gethostbyname(host, &addr);
build ok,
mbedtls_ssl_handshake no error
mqtt_init no error
mqtt_connect no error
but server not real connected;
mqtt_sync/mqtt_publish/mqtt_sync return MQTT_ERROR_UNKNOWN

Can you help me solve this problem?